prusa3d/Original-Prusa-MINI #1: M600 / Filament change

Kasperc88, 2019-12-20:
There is two issues with the Original Prusa MINI and M600 (Filament change) that can be fixed with firmware.

Issues:

1#
When the printer unload the filament it will turn off the nozzle. This is fine. To load the new filament you have to press continue on the LCD,
and the printer will now heat up the nozzle to the target temperature. Once it hit the target temperature it will begin loading the filament.

This mean you have to sit and wait while you push the new filament into the tube so the printer can grab it. If it was not grabbet from the start
the filament will not be loaded correct and this lead to issue #2.

#2
After the printer thinks it have loaded the filament it will ask you if the color is correct. You can press "No" and it will extruder a small amount
of filament. But after this it will start to print. You do not get a new chance to say the color still isn't correct.

Fixes:
#1 - A option to tell the printer to begin loading the new filament
#2 - More chances to tell that the color isn't correct.

Bonus: 
It could be helpfull if the z-axis raise alot more so we can clean the nozzle without burning our hands.

joeynuggetz, 2019-12-24:
Agree with the filament loading/unloading especially during a color change. Just got a ruined print because the filament didn't load on the first attempt (extruder gears didn't grab filament). Instead of asking if filament loaded just like any other printer it just assumed everything was fine when I said "no" to the color ok message. This needs to be fixed asap as color prints will fail if there are any problems encountered while loading new filament. 

I do not have the filament sensor. Not sure if having one would have helped in this case.

charlesramsay, 2020-01-04:
I have found it impossible to unload PETG several times.  The unload routine does not allow the filament to be completely released.  I came up with a solution.  I cut the strand from the spool. Then I purge the remaining filament until it's beyond the stepper motor. Then I undo the bowden tube at the stepper. It works.

JohnnyDeer, 2020-06-15:
Hi @Kasperc88 , this issue is related to Prusa-Firmware-Buddy. Is duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/260 and https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/361 so i´ll close this one. Thank you for suggestion and understanding.

closed 2020-06-15 by JohnnyDeer