prusa3d/Original-Prusa-MINI #4: MINDA cable wire break (mesh bed leveling unexpected behaviour/does not proceed)

6ffm70, 2019-12-26:
**Description**: 4x4 point mesh bed leveling does not finish as expected. When process starts, MINDA light is off; 
probing pos. 1 - the print head moves _up_ (couple of mm)
probing pos. 2 - the print head moves _up_ (couple of mm)
probing pos. 3 - MINDA light switches on, print head moves _down_, then up; here the process halts.

When moving the X-axis, the MINA light switches on in a position further away from the extruder; and switches off approximately at the same position when moving back into direction of the extruder.

This behaviour is reproduceable. With every go, the Z-Axis moves further up by a couple of mm.

**Analysis**: with the printer switched on, the MINDA light on or off, moving the MINA cable by hand makes the MINDA light flicker.

**Assumption**: at least one of the conductors in the MINDA cable is broken after ~30 hours of operation.

6ffm70, 2019-12-29:
**More Analysis**: locating the wire break in the cable
1. MINDA and cable in situ: removed textile sleeve, singled out MINDA cable
2. marking (I) the position of the top cable tie on the cable for reference (28.5cm from MINDA):
3. bending the cable starting at the sensor, moving in direction of connector end while watching the MINDA light
4. marking (II) the first position where the light flickers first (24cm from MINDA)
5. switching off printer, removing connector and cable
[image]
6. clean cutting cable on marking II in direction of MINDA
[image]
7. removing outer cable 15mm and wire isolation 7mm from cut of connector section half
[image]
[image]
8. measuring each resistance of all three wires while bending the cable in different sections: black OK, brown OK, blue NOK
9. locating the approximate point of non-contact in the blue wire by watching the multimeter and bending the cable, starting from the connector; marking (III) the approximate point (39cm from MINDA)
10. carefully stripping blue wire isolation from cut into direction of marking III
[image]
11. noting down position where stranded wire is separated (**29cm** (IV), **33.5cm** (V) from MINDA)
12. checking the remainder cable length for breaks with resistance measurement

6ffm70, 2019-12-29:
### There is no evidence for the root cause of the wire break so far. As long as the root cause is not eliminated, any replacement of the MINDA ass'y or repair hack is a temporary solution.

**Repair hack**:
Parts required: 5x cable ties 2.5mm, 4x wire end sleeves*, heat shrink hose ø3mm, ø6mm as required
Tools required: wire stripper 4mm², 0.25-0.4mm², copper wire shear, end sleeve crimp pliers (open type) 0.5mm²*
*alternatively a heat controlled soldering iron and soldering wire

1. clean cut off section with broken wire strands
2. stripping shortened blue wire for 7mm
3. preparing new isolated (yellow in pictures) wire in length of broken/removed wire with 7mm isolation stripped of both ends
4. crimping (to avoid uncontrolled heat) end of blue wire with one end of new wire with an end sleeve
5. isolating wire junction with heat shrink hose
6. removing outer cable 15mm and wire isolation 7mm from cut of MINDA section half
7. preparing 3x heat shrink hose over wire end, and 1x heat cable sleeve replacement shrink hose over cable in length of stripped section +30mm
8. repeat crimp connection for 3x wire connection
[image]
[image]
9. isolate 3x wire junction with heat shrink hoses
10. heat shrink cable sleeve replacement hose
[image]
11. place MINDA assembly back in printer using new cable ties
12. hope that changed internal wire resistances don't affect sensor operation

gregsaun, 2020-01-07:
The hotend cables on the Prusa Mini are very short and stressed which could potentially have damaged your wire quicker than expected.

I made a fix for those too short cables in case you want to solve it:
https://www.thingiverse.com/thing:4065410

6ffm70, 2020-01-14:
More follow up:

1. The cable as repaired above made the printer go again; however Mesh Bed Leveling gave different results by +/-0.400 each time, resulting in unsuccessful prints.
2. Replacing the repaired MINDA assembly with a new one provided by P/R support resolved this issue.
3. The cause of 1.) is unknown to me. Assumptions: one or both of the wires not repaired were damaged, and this was not detected during multimeter tests; or the repair affected sensor operation.

JohnnyDeer, 2020-06-16:
Hi @6ffm70 , we sended replacements with new MINDA type and improved cable holder, wich solving this problems. If you still have this kind of MINDA issue please contact our Tech Support. Thank you.

closed 2020-06-16 by JohnnyDeer

6ffm70, 2020-08-01:
Hello @JohnnyDeer, can you please elaborate on what is meant with _new MINDA type_? I've had a couple of MINDAs replaced by tech support, and they all showed similar symptoms, even with the improved cable holder.

6ffm70, 2020-08-26:
Follow up: new MINDA details added here: https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/ELECTRONICS/mini-minda.pdf