prusa3d/Original-Prusa-MINI #5: Mesh Levelling has very weird behaviour

AndyThirtover, 2020-01-07:
4.0.1 Please view:

https://www.youtube.com/watch?v=rOWLI0K9chY

You'll see that the points 1-2 work perfectly and the 3-4-5 only go UP.  6-7-8 are ok and and that the MINDA does not operate.  Notice that 6-7-8 are ok and then 11-12 and 13 also go up (as do all of the points on the left hand side of the bed.

The net result is that this printer cannot be first layer calibrated.   It seems that the issue gets worse every time I power cycle/factory reset the printer.    I've managed one print on 4.0.0 but since the 4.0.1 upgrade this sad Prusa Mini has not managed a single print ;)

gregsaun, 2020-01-07:
@michalxfanta I suspect this is due to the too short cables on the hotend which wear the cables quicker than expected. I have made an improvement on your parts to help, you can find it here: https://www.thingiverse.com/thing:4065410

Is Prusa working on a solution for this issue? The best way I see today would be to provide a longer heater cartridge cable because this is the shortest and most problematic one.

JohnnyDeer, 2020-06-16:
Hi, this behavior is because cables of MINDA are broken. We sended replacements with new MINDA type and improved cable holder, wich solving this problems. If you still have this kind of MINDA issue please contact our Tech Support. Thank you.

closed 2020-06-16 by JohnnyDeer