prusa3d/Original-Prusa-MINI #11: Stopping a print causes the printhead to crash into printed part.

HenrikPankoke, 2020-03-09:
I had a print with 6 elements printing individual.
The elements is place like the photo. print is from back left to back right.
moving to front left to front right
[image]

When printing the front right element I stopped the print.
the printhead then moves backwards an into the element behind. causing this to be knocked of the plate. In this case it was a small part and not theat high adhesion. But larger parts with could adhesion could have damage that fan on the printhead. 
[image]

maybe changing stop command to move the nozzle up above highest printed element, before moving in any other direction. or just move nozzle clear of current print and stop movement.

HenrikPankoke, 2020-03-09:
Sorry. created this the wrong place. Don't know how to delete.
Moved to https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/257

3d-gussner, 2020-04-06:
@HenrikPankoke you find a `Close issue` button below ;-)
[image]

JohnnyDeer, 2020-06-15:
I´m closing this issue as duplicate.

closed 2020-06-15 by JohnnyDeer