prusa3d/Original-Prusa-MINI #15: Prusa Mini Factory USB Stick failing

feutl, 2020-04-24:
Hello,

after some people do report the same problem on the Prusa Forum, I thought I should let you know.
The USB Stick shipped with the Prusa Mini fails after some time being mounted. I was happy the firmware update went through, but every print on the USB stick failed.

Copied the files to another USB stick and everything worked straight away.

https://forum.prusaprinters.org/forum/hardware-firmware-and-software-help/factory-usb-drive-wont-stay-mounted-on-printer-crashes-failed-prints/
https://forum.prusaprinters.org/forum/assembly-and-first-prints-troubleshooting/watchdog-reset-2/

3d-gussner, 2020-04-24:
Hi, I think you posted in wrong repository.

feutl, 2020-04-30:
> 
> 
> Hi, I think you posted in wrong repository.

which one would be the right repo ?
I experienced this with the Prusa MINI - and the forum seems not the right place at all

abosson, 2020-05-21:
I had the same problem with the USB stick... Dôme " watchdog reset" I change the USB stick and copy thé file without reslicing and I have no problem tête!

JohnnyDeer, 2020-06-15:
Hi @feutl , this issue is related to firmware and is duplicate of https://github.com/prusa3d/Prusa-Firmware-Buddy/issues/255 . I´ll close this one.

closed 2020-06-15 by JohnnyDeer