prusa3d/Original-Prusa-MINI #21: OpenSCAD?

infosisio, 2020-05-20:
Is there a particular reason that this wasn't designed in OpenSCAD as per the usual tradition?

Thanks.

JohnnyDeer, 2020-06-15:
Hi @infosisio , reason why we are no longer using OpenSCAD is that we found better solution.

infosisio, 2020-06-15:
@JohnnyDeer I don't know what solution you found but you are no longer sharing them in a format the community can build upon.

JohnnyDeer, 2020-06-22:
@infosisio we are still sharing all data in format which is usable to build upon.

infosisio, 2020-06-22:
@JohnnyDeer That is not true, STEP is an archival format that doesn't give parametric control. The only thing it is useful for is to act as reference for reverse engineering a part, in order to be able to make changes and improvements. Please reconsider this.

SyberxSpace, 2020-06-22:
From the headers in the released STEP files it appears they were generated using Autodesk Inventor. Example from MINI-x-end.stp:
```
FILE_NAME(
/* name */ 
'MINI-x-end.stp',
/* time_stamp */ '2020-01-22T16:21:13+01:00',
/* author */ ('Robert Turinsky'),
/* organization */ (''),
/* preprocessor_version */ 'ST-DEVELOPER v17.2',
/* originating_system */ 'Autodesk Inventor 2019',
/* authorisation */ '');
```
I believe this would mean that the STEP files are indeed not the source files, and that this would make the printer not open-source as claimed. This also severely hampers the speed and accuracy of community improvements. Even if you Prusa Research believes that STEP files are source files **(please make clear if this is the case)**, I think it can be agreed that the practice is against the community accepted definition of the term and how it is used to imply an ecosystem encouraging communal improvement and iteration.

CacikeOtoao, 2020-06-22:
Sadly this has been happening more often that a lot of us would like. They have been dragging their feet to release source files or just setting their own timelines and borderline redefining the licenses used to release their products. They have actually made changes to some of the “Open Source” products and not update the git files until been pressed on. @JohnnyDeer  it is not about releasing “usable files to build upon”, it is about releasing the actual source files for products that are touted as Open Source and come with a big Open Source logo stamped on the box.

infosisio, 2020-06-24:
@JohnnyDeer I am sorry to say this but we are not talking to a wall here. You've heard our points clearly and you owe us a response. You market something as open source when it's not. Please explain yourselves.

CacikeOtoao, 2020-06-24:
I will also like an explanation, hopefully this is not a case of #openwashing.

lord-carlos, 2020-06-24:
It's a bit like releasing a binary Java file and calling it open source because you can decompile it.

It's nice that Prusa is releasing the step file, but calling it an open source printer at the same time is appalling.
On the shop site it even says:
> All parts of our printer are open-source.

infosisio, 2020-06-26:
@CacikeOtoao Unfortunately so far this seems to be the case...

josefprusa, 2020-06-30:
Let me clarify and correct some statements here from our side.
We have used openSCAD for long time, but from certain moment we needed to switch to more advanced sw for several reasons - starting from increasing speed of development, documentation, more advanced features available and ending up with easier exchange of data with our suppliers.
For GIT we choose STEP file format as this is widely used for exchange of data ( STEP = Standard for the Exchange of Product model data) and in fact it has been requested often by developers who did not like openSCAD in MK3 repository as alternative. Also, STEP files are widely accepted by various CAD software, it is standard how to exchange CAD information but   I do agree that STEP is not ideal for later development as you do loose some functionality or information as discussed here.
Starting with MINI repository we will add native source files in addition to STEP files ( we now use Autodesk Inventor ) and missing pieces of documentation in short time.

infosisio, 2020-06-30:
@josefprusa Thank you for agreeing to provide the source files but allow me to say as a response to your argument that STEP is not sufficient for an open source project, it is simply not the source file. Thanks.

SyberxSpace, 2020-07-01:
Thank you @josefprusa for agreeing that native source files are important for the community to develop and improve in an open and efficient manner. I'm hoping this will also carry over to the mk3s parts that also lack source files at the moment.

In the future you might want to consider a different license on release and make it clear what your release schedule and intents are from the beginning. "Surprises" like the appendix being released post-preorder and the lack of files and firmware tools aren't good for the community or how your brand is seen by the community. Not too long ago you were an outstanding example of what was good in the open community and these recent evens have begun to tarnish that. I don't think it's too late to turn that around though!

TLDR: The community is smart enough to understand if you just tell them what you plan outright. Just be **open** about it and stay faithful to the true intent of the open-source practice if you really want to be called open-source. Don't just do the minimum to keep the open tag on your products as it just makes you look untrustworthy.

I'm excited and hopeful for the future!

ucirello, 2021-02-23:
Prusa can keep the Autodesk Inventor files - as long as they release a version in openscad (or freecad for that matter) that the community can open and modify at source level, not at archive level (like Step or STL).