prusa3d/Original-Prusa-MINI #34: Prusa Mini Open Source Design

6ffm70, 2020-08-01:
Seeing the good attitude towards open source hardware, software and mechanic design at Prusa Research marks a new era for me. An innovative, fast growing company that benefits from sharing, and lets others benefit from sharing:
[image]
Looking at the mechanic design in the repositories here, I have the impression that the open source design promise is not met, when I compare this to the creation of software: everything is available to modify, build and test it to a highly productive level. 
External people, individuals and professional contribute, and have the opportunity to try new features. 

Not so on the mechanic side - this is pretty much closed to Prusa internal and largely undocumented and incomplete. If traces are to be found, they are scattered over different sites in different categories. This is a missed opportunity to let P/R and the community benefit from open-source.

6ffm70, 2020-08-01:
To have this set up, I'd like to see

- 3d design definitions that can be modified. Unlike #21 , I don't think that has to be a particular file format like openSCAD. Inventor and STEP files should be okay, as people used different CAD systems to contribute. Though, if external references or skeleton models are used, this should stay intact within the native CAD files.
- Product and component specifications: the requirements for the entire product should be broken down to component level - including engineering drawings or model based definitions. Test cases and criteria, tolerances, tolerance calculations, and where applicable D-FMEA.
- If there are third party components that are not open source, they should include a datasheet and a customer volume model; if those parts are customer (Prusa) directed, like some E3D components on the Mk3, the specifications and all other open source data should be included too.
- Production data - the saying goes that Prusa printers can reproduce large parts of themselves (for all 3d-printed components), yet, there's only STL files in the repo, and no printing parameters. I'm with #18 here - why keep the -no doubt- super refined and experienced printing parameters P/R internal? The 3mf files, and the material properties should be open too.
- Everything in one place: an example: I only learned by chance that Mk3 and Mk3s use different PTFE tube diameters by tenths of a millimetre. That information was buried deep in support articles; even though it's relevant to function of the printer, and I was troubleshooting mine. I see that Github is more limited to version 3d designs while maintaining file relations, than a PDM system. Maybe enhance prusaprinters.org to host community design projects, and showcase this function by maintaining all engineering data for the printers there?
When this is properly set up, I'm sure external designer will branch out designs, test, enhance, etc. The current quality of sharing is discouraging this, to be honest.

gregsaun, 2020-08-01:
I think you made a good summary of the situation.

However, I am not okay with STEP file being a source. Even for designing a simple cube you have to start with a sketch and some defined dimensions and constraints (90° angle for example). If you export as STEP you loose this sketch and so part of the logic behind the design. Note that the dimensions and constraints are not the only logic you are loosing but also the sequence used to design a part. Some parts can be fairly complex to design and requires some special steps.

Just a side note regarding MK3(S) printers, they have never been sold as open source as far as I remember. Nothing was ever mentioned in this sense on the prusa3d.com pages nor in the launch video. It was just made well enough to think it is an open source printer but it is not. The Prusa Mini is different as it has been claimed in several places to be an open source printer and they even mention "All parts of our printer are open-source" (and not _all printed parts_).

6ffm70, 2020-08-01:
@gregsaun, this is grabbed from https://www.prusa3d.com/about-us/ , and this is what I also mean by good attitude towards open source hardware - it covers everything:
[image]
The fragments of the i3 Mk3/s design published in the github repo have the GPL licenece attached as well as the MINI.

I don't think the _sources_ of the mechanic design are deliberately incomplete. **Lack of an established method to collaborate in public may be a cause for this.**

This is also where STEP (subject to a lot of discussions eg. here https://github.com/prusa3d/Original-Prusa-i3/issues/139, https://github.com/prusa3d/Original-Prusa-i3/issues/84#) comes in as an _addition_ to the native format - it opens up editing access to the highest possible share of the community.

The tool of choice for the main branch will be set by the main developer; yesterday it was openSCAD, today it's Inventor (or some that identifies as Inventor), tomorrow it's another solution that fits the development requirements. 

I think your concerns about the STEP format are valid; while at the same time my issue is the completeness of the _source_ of all mechanical engineering aspects of the printers. 

Example: even if all native parametric 3d files were published online, you'd still only see how they were drawn/modelled, but not how the whole system works, or which design features or tolerances/materials have been chosen to overcome e.g. a heat dissipation challenge. You'd not be able to build on this knowledge to enhance this feature.

When calling the whole product, thus also the mechanical engineering _open source_, the community should be able to engineer the whole printer from scratch to make the project benefit from its contributions.

gregsaun, 2020-08-02:
@6ffm70 I agree for the STEP file, it is important to me to share them if you want to share properly your project. However they are not the source. You can create a STEP from the source but can't create the source from the STEP file. GPL is clear, you must share the source, it is not optional.

I agree with you that a CAD files alone isn't valid, you need to have the drawings for the tolerances/material/surface finish for "metal" parts and the print settings for the printed parts.

Good point regarding the about-us section. Then a lot more is missing like the specs of the fans, the specs of the extruder spring, the magnet specs, the whole BOM, etc. And this is valid for MK2(S) and MK2.5(S) as well (and probably MK1).

_> I don't think the sources of the mechanic design are deliberately incomplete. Lack of an established method for open collaboration may be a cause for this._
I think the community here starts to see this differently now. It is really not the first time we ask for the source files for MK2(S), MK2.5(S), MK3(S) and MINI. They had many many chances to fix this.

EuphoricPenguin, 2020-08-25:
I think the recent additions might push this thread closer to being closed, but there's still a distinct lack of community involvement on the hardware side; at least we have more or less complete documentation.

6ffm70, 2020-08-26:
@EuphoricPenguin , I agree, this is a leap forward!

For lack of finding a standard for open collaborative mechanical engineering, I started a document to collect/compile what's important to the community: 

[image]

This document is maintained here: [https://github.com/6ffm70/opensource_mechanicalengineering](url)

JohnnyDeer, 2020-11-19:
Original Prusa MINI is now certified by Open Source HW Aliance - https://certification.oshwa.org/cz000002.html
I believe this issue can be closed as well as #21, @6ffm70 ?

lord-carlos, 2020-12-08:
> Original Prusa MINI is now certified by Open Source HW Aliance - https://certification.oshwa.org/cz000002.html
> I believe this issue can be closed as well as #21, @6ffm70 ?

That is only for software, no?
Hardware is "other"

ucirello, 2021-02-23:
By effectively forcing the use of a closed source software for the original design, this printer should not be labeled as open source hardware. Especially, because Autodesk Inventor is a very expensive software. By using an expensive proprietary software, Prusa is effectively locking contributions and fork out of this design.

6ffm70, 2021-03-02:
> > Original Prusa MINI is now certified by Open Source HW Aliance - https://certification.oshwa.org/cz000002.html
> > I believe this issue can be closed as well as #21, @6ffm70 ?
> 
> That is only for software, no?
> Hardware is "other"

That's true - there is no licence for the hardware; and no licence for the documentation; just links to the github repo. 
@JohnnyDeer can you please be more specific as to what has been certified?

bomarspub, 2021-03-17:
I found this to me an interesting read. I am currently reviewing the mechanicals, and several things are puzzling. I get it that its a volunteer effort, but maybe a little mechanical drafting guidance is in order. One thing is the absence of a preferred vendor part number is absent from the drawing.  Two  The mechanical call outs seem to be all over the map and need more explanation to the intention. Possibly a legend for international sharing to make more sense out of them. Cases where bubbles indicate there should be a second sheet or a defined note, yet absent. Just using the rods as a simple example, X rod is missing the rockwell spec.  Certainly can purchase this over the counter , but it would be nice to have a preferred vendor to work with. my two cents..

pltaylor, 2021-05-27:
> By effectively forcing the use of a closed source software for the original design, this printer should not be labeled as open source hardware. Especially, because Autodesk Inventor is a very [...]

It may not be in the true spirit of open source, but asking Prusa to hamstring themselves by forcing the use of open source design tools may be a bridge too far.  At the end of the day Prusa is a business and they need to use the tools they think will help them get the job done.  Every step towards a pure open source is commendable, but like everything else in engineering there has to be a balance of multiple factors.

ucirello, 2021-05-27:
@pltaylor they can keep the originals in a closed format, and release the source in a open format that actually has the same object tree (freeCAD comes to mind, but not only). No need to hamstring themselves - but take the effort of saving into more (and open) formats.

pltaylor, 2021-05-27:
But if you submit a PR they would have to then do a manual merge into their preferred design software.  If the goal is to make modifications for yourself, then yes, your suggestion works.  But if the goal is to actually collaborate and grow the platform then your suggestion doesn't scale much better than the current state of things.

ucirello, 2021-05-27:
So just declare Mini to not be open source and be done with it.

6ffm70, 2021-05-29:
> But if you submit a PR they would have to then do a manual merge into their preferred design software. If the goal is to make modifications for yourself, then yes, your suggestion works. But if the [...]

I can see your point, and I believe it is a description of the ideal situation - which should certainly be the goal.
At the same time, the software available for collaborative mechanical design (PDM and CAD) is excellent for productivity and maintaining references, file associations and design intents; however none of these is open source. 

This leaves the STEP format as the lowest common (open) exchange format, for the price of abandoning design parameters and intent, but maintaining accuracy. For now, the best every creator of opensource design data can do, is publishing STEP and native formats and what is available from the aforementioned (https://github.com/prusa3d/Original-Prusa-MINI/issues/34#issuecomment-680780459) documentation.

There is clearly a gap in tools for online open collaboration in mechanical design as compared to software development, and Prusa obviously does not take advantage of merge requests for mechanical design in this and the other github parts repositories; though increasing the amount of data in this MINI repo has been a step in the right direction.