prusa3d/Original-Prusa-MINI #42: why can not connect mini to pronterface

Surealjedeye, 2021-02-02:
I have been trying to figure this out. I have easily connected 4 other branded printers to pronterface via usb. I connect the mini to my computer through usb. The computer recognizes it on Com 7. No matter the baudrate selected pronterface can not connect. I have tried everything me and my son can figure out. I just ended a chat with Prusa customer support and I was referred to this forum. Please help. I am so frustrated at the moment.

BogdanH59, 2021-05-25:
On Windows 10 I can connect Pronterface to MINI (I use baudrate 250000). Did you use mini USB port on MINI?

DRracer, 2021-07-30:
@Surealjedeye is your issue still relevant? Have you tried the latest FW 4.3.2-RC2?

zoltanbaranyai, 2022-12-15:
I have the same issue, any solution? Latest FW on printer and installed the latest driver from the prusa site.

Prusa-Support, 2023-01-03:
Hello. As mentioned in the Handbook, the micro USB port allows further third-party applications and custom uses.

> The micro USB port can be used to run third-party solutions

https://cdn.prusa3d.com/downloads/manual/prusa3d_manual_mini_en.pdf

This means that as of now, the use of it is not directly supported by Prusa Research. In fact, I'm afraid our drivers pack doesn't include any specific drivers for the correct functionality of the Prusa MINI via computer.
A user suggested some specific drivers in the Forum. Feel free to test this suggestion or take part in the discussion.
https://forum.prusa3d.com/forum/general-software-discussion/mini-and-pronterface-not-connecting/#post-489157

Other than this, apparently, there are two types of micro-USB cables on the market: charge/power-only cables and data/sync cables. The difference is apparently the power transmission capacity and the number of threads inside. You may want to make sure to use a cable suitable for data transfer.

Michele Moramarco
Prusa Research

DRracer, 2023-01-04:
Standard serial line is available on the MINI without limitations, but:
- baudrate needs to be set to 115200 bd
- you really need to use a data/sync micro-USB cable
- I'd recommend using firmware 4.4.0 and/or 4.4.1-RC1. There have been some issues with the serial line in the beta releases before 4.4.0

FYI: opening the serial line at 57600 bd shall be used for debugging - it makes the MINI send Marlin logs. This way you cannot issue G-codes into the printer.

The best way of checking whether the MINI connects to a computer is to look into the operating system logs and search for something like:
```
kernel: usb 1-1.2: New USB device found, idVendor=2c99, idProduct=000c, bcdDevice= 2.00
kernel: usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
kernel: usb 1-1.2: Product: Original Prusa MINI
kernel: usb 1-1.2: Manufacturer: Prusa Research (prusa3d.com)
kernel: usb 1-1.2: SerialNumber: CZPX.............
kernel: cdc_acm 1-1.2:1.0: ttyACM5: USB ACM device
```
This confirms, that Linux "sees" the new serial device - the MINI. From now on the serial port `/dev/ttyACM5` can be used to talk to the printer.

SimonTratter123, 2023-01-07:
I am able to use Octoprint with an Raspberry Pi with the micro USB port on the Mini. However I can't connect the mini to my computer to use Pronterface. Devicemanager does not find any device refering to the mini.

Prusa-Support, 2023-01-19:
Unfortunately the drivers are not included.
Please try the drivers suggested by the community on the forum.
Also make sure the micro USB port is not damaged in any way (wobbly, enlarged, detached or dry soldering...).

Michele Moramarco
Prusa Research

hkienle, 2025-02-12:
> micro USB port

Just to emphasize, because this may be easily missed: The USB-A port will not work, you have to connect to the USB Micro port that sits to the left of the USB-A port.

[image]
Source: https://help.prusa3d.com/article/octoprint-configuration-and-install_2182