prusa3d/Original-Prusa-MINI #45: USB stick occasionally becomes unreadable.

damiandixon, 2021-04-09:
Firmware: 4.3.1+2759

USB stick (one supplied with printer) is ejected correctly from Windows but occasionally becomes unreadable on insertion into printer and subsequently still unreadable on re-insertion into windows laptop.

Happened twice now. Both times reformatted.

raymate, 2021-04-09:
Try another stick, may not be the firmware. A few users have reported the supplied stick becomes bad or not work well. I didn't even bother using it. I got myself a Kingston branded one. DTSE9, I have two versions one 32GB and an older 4GB. The 4GB work without any issues. The 32GB sometimes does not show up in the printer. This is what he latest firmware might fix as its USB driver related. Most say use a stick no bigger than 8GB. I tend to agree with current stable firmware. But getting hold of small USB sticks is becoming hard, I looked and I would need to resort to eBay to get a 4GB stick now days. So I hope once this latest firmware becomes live it will let me use my 32GB stick with no issue.

damiandixon, 2021-04-21:
A new 4GB USB integral drive seems to be working fine.

closed 2021-04-21 by damiandixon