prusa3d/Original-Prusa-MINI #47: Problems with Hostnames in firmware 4.3.1 (Using more than one Prusa mini)

waltercwitt, 2021-04-21:
Repost from prusa printers forum
"
Hello,

I've been chasing this issue around for quite some time now, but it appears that no-one else is having a similar issue. At work, I've setup 2 Prusa MINIs with both of them connected to our network via the same ethernet switch. I set them up with the hostnames, http://prusamini1 and http://prusamini2 to make it easier for people to remember where to check the status of there prints.

The problem I'm having is, sometimes after restarting the printer, the hostname gets wiped, and the printers can only be accessed by there IP address, and in rare cases, one of the printers will take the hostname of the other. This was actually happening a bit before firmware 4.3.0, but it seems to have gotten far worse after updating them.

During my troubleshooting of the issue, I discovered some very odd things that the printers where doing:

- If, during power on, there was a flash drive that had a lansettings.ini file on it plugged into the machine, It would overwrite it's current settings with the ones in the file. (I'm pretty sure this is what was causing one printer to take the others hostname)
- When I save the LAN settings to a flash drive, the hostname it saves is always blank, even when the hostname is set and working. (I am not allowed to set as static IP for the printers at work, so I can't verify if any of the other settings are returning default values as well.)
- When I load the LAN settings from a flash drive using the menu, it very rarely actually works, meaning I have to use the bootup method in my first point to fix it. (Again I have only ever tried changing the hostname, all the other settings I have to leave as default as per my works IT network policy.)
 
Aside from that, I'm finding it quite annoying that the printer will display every other network setting in the LAN settings menu, EXCEPT the hostname, so in order to check if I've successfully fixed it, I have to walk all the way back to my computer on the other side of the building to test it.

I've made sure that I don't have a lansettings.ini file on any of the flash drives I use, but they will still sometimes just forget the hostnames anyway. and then I have to save a new file to a flash drive, modify it on my computer, turn off the printer, and boot it up with the flash drive inserted to fix it, and sometimes, it doesn't work the first time.

Has anyone else had similar issues with there printer forgetting LAN settings since the 4.3.0 update?
"

okaerin, 2021-04-25:
@waltercwitt This sounds rather unpleasent. Unfortunately this is the wrong repository. It only contains the hardware/design parts. The firmware of the prusa mini is maintained at [https://github.com/prusa3d/Prusa-Firmware-Buddy](https://github.com/prusa3d/Prusa-Firmware-Buddy). I would suggest that you close the issue here and post it over there [https://github.com/prusa3d/Prusa-Firmware-Buddy/issues](https://github.com/prusa3d/Prusa-Firmware-Buddy/issues) . I hope that the prusa people maintaining that can figure that one out.

waltercwitt, 2021-04-27:
Oh, sorry, I was sent here by a Prusa support person. I'll do that now

closed 2021-04-27 by waltercwitt