prusa3d/Original-Prusa-MINI #48: USB Hub not working

passteinforth, 2021-04-23:
Hi,
I want to use a USB Hub to connect more USB devices to the Mini. I use this Hub https://www.alternate.de/i-tec/USB-3-0-Metal-HUB-4-Port-USB-Hub/html/product/1524373 and connect it to the USB-A Port of the Mini. The little LED of the Hub turns on, but when connecting the original USB stick to load print files, nothing shows up and in the main menu it still says 'no USB'. The USB stick is the only device plugged into the hub at this moment.
When connecting an Arduino Nano to the USB hub, it gets powered as intended (the Nano controls a humidity sensor for a drybox and a 7-digit displays and only draws some power, no data connection). 
The second USB port of my Mini is blocked by Octoprint/Raspberry Pi 4, so I cant use this to power my humidity sensor/Arduino.

JohnnyDeer, 2021-05-17:
Hi @passteinforth, USB hub is not supported by MINI, this won't work even if you plug in just one USB stick. I´m closing this issue as invalid.

closed 2021-05-17 by JohnnyDeer

aargonz23, 2022-03-24:
@JohnnyDeer @josefprusa why wont the mini+ support a hub? I tried a switcher and that wouldnt work either. why not have it wirelessly connect? arent there more options for the other printers? im not sure I understand the issue with setting up the mini to be more convenient than just swapping the USB drive out constantly. I know there are options to connect a raspberry Pi to a mini but they're not exactly easy to come by this year. I've tried and they sell out before you even think to check. I know its not a massive issue but better options would be massively appreciated by the mini+ community as a whole.

JakoobCZ, 2022-04-06:
Hi guys, the MINI+ USB port was designed to support a USB flash drive. Powering any other external device (like USB hub) was not planned nor tested. Getting the USB hub to work including full power would require HW adjustments to the board and changes to the firmware.