prusa3d/Original-Prusa-MINI #50: Lead Screw not called out on Z-Axis BOM

pltaylor, 2021-05-27:
The specs for the lead screw are missing from the Z-Axis BOM on https://github.com/prusa3d/Original-Prusa-MINI/blob/master/BOM/Z-AXIS.md

JohnnyDeer, 2021-06-22:
Hi @pltaylor, thank you for your comment. We will add specs of lead screw to X-Axis BOM where is secured trapeze screw.

Prusa-Support, 2022-05-10:
Hello! Sorry for replying so late to this thread. The Z-axis motor leadscrew specs are currently within the "MINI Z-axis motor" link, which points to the documentation of all the MINI motors: https://github.com/prusa3d/Original-Prusa-MINI/blob/master/DOCUMENTATION/ELECTRONICS/mini-motor-kit.pdf. Within that page, there is the Z-axis motor documentation, which includes the leadscrew and trapezoid nut specs. 

Alessandro Pantaleo
Prusa Research

closed 2022-05-10 by Prusa-Support