prusa3d/Original-Prusa-MINI #58: [ENHANCEMENT] Dual Gear extruder

SnickCH, 2021-12-26:
It would be great, if the Prusa mini/mini+ would receive an extruder upgrade with dual gear. There is a reason that Bondtech can sell many upgrades for the Prusa mini and even the community on Thingiverse is providing new extruders for the mini. With an official extruder it would make our life much easier.

The main problem with the actual extruder are the following in my opinion

- soft filament can't be printed, because the gears will rasp the filament and cause that it will be stock in the extruder
- the gear teeth are to fine /small
- Not all kind of flex Filament can be printed
- precision could be improved
- Hard to clean. It must be disassembled to be cleaned. Filament that was rasped gets stuck inside 

With eSun and Prusament I didn't have any problems. As soon as I used a softer filament, I had to clean the whole extruder (and also the hotend, because of the clog that was produced). If you compare the price for a dual gear extruder it would not cost much more if it comes on the stock printer. If you have to upgrade it later, it is an expensive task. 

Example: https://www.thingiverse.com/thing:4179460

Prusa-Support, 2023-08-25:
Thanks for your suggestion.

The printer is Open Sorce, hence easily modifiable.
Officially we can't offer direct support for mods but we love to see the active members of the community sharing projects and ideas and surely we won't ignore them.
As far as I know, there aren't new model iterations in the current plans but we are listening.

Michele Moramarco
Prusa Research

KlausKraemer, 2023-08-27:
Hi Michele,
it's not about easy modifying - it's about quality!
Best regards 
Klaus