prusa3d/Original-Prusa-MINI #59: [ENHANCEMENT] Live Z option while printing at print screen

OopAck, 2022-02-10:
Requesting a Live Z option while printing and at print screen to the firmware.

Martenz, 2022-05-17:
Do you mean to have also the Z level number together with the Z level in mm on the lcd screen? If so I totally agree, I found I use the printer with usb most of the time and I'd like to know which level is printing.
Something like percentage of the print and level printing:

83% - 272/326

Prusa-Support, 2022-05-17:
Hello! Thank you for the request. @OopAck can you please expand on the request? Did you mean the Live-Z adjust in the format -0.123 or more the percentage as @Martenz mentioned? 

If it was the first case, was there an issue with inconsistent Live-Z adjust values throughout the same print or across prints? Let me know in that case, we can expand on it. 

Alessandro Pantaleo
Prusa Research

OopAck, 2022-05-17:
Figured out.

closed 2022-05-17 by OopAck