prusa3d/Original-Prusa-MINI #63: Prusa Mini+ Z-top R1.1 - Extra Height

seifer69er, 2022-07-24:
Based on 3d file uploaded to PRUSA Printers,
https://www.printables.com/model/202467-prusa-mini-z-top-r11-extra-height
and
https://www.printables.com/model/179537-prusa-mini-z-axis-extra-height-adaptor

There is an issue that the extruder assembly owls on the Z axis top.
This in turn is not the true end stop of the PRUSA MINI+.
By removing material of the z top to allow the extruder to pass through, this in turn allows the prusa mini to gain an additional ~4-5mm in z height.
[image]

Prusa-Support, 2022-07-25:
Hello! Thank you for your feedback. If I understood correctly, the issue was only with the parts in the link, but these are not our original parts - our original MINI/+ printable parts are in this folder: https://www.printables.com/model/57214-mini-printable-parts, would the issue be there also with our original parts? 

Let me know, in case of anything to further review, do not hesitate to ask. 

Alessandro Pantaleo
Prusa Research

seifer69er, 2022-08-01:
Hi there, yes, the issue is with the original parts, the Z top will need to be trimmed as per the link attached R1.1.
I can provide the step file or any additional clarity as required,

seifer69er, 2022-08-02:
Hi there,

That is correct, a minimum of 5mm additional z axis height is achieved

Prusa-Support, 2022-08-02:
Thank you for the confirmation! OK, tagged it as enhancement, no guarantees it would be implemented, and likely it won't, but will be there for devs to further review in case. Thank you once again for the suggestion. 

Alessandro Pantaleo
Prusa Research