prusa3d/Original-Prusa-MINI #67: MINI+ Suggestions/Enhancements: time remaining option, menus, bed heating, purge, PrusaLink

re490, 2023-01-31:
MINI+ Wish List / Suggestions:

1) 4.4.1 Firmware changed to showing Clock time for print finishing;   Consider option to toggle print time between clock time or hr/min remaining   (I prefer knowing hrs/min remaining, not clock time)

2) Make Front knob rotation consistent in Menus:  in Menus, clockwise moves selection down, but in adjusting Z-height, clockwise moves the nozzle up.

3) Why does Heated Bed need to heat up when loading/unloading filament?    (doesn't save print time)

4) during filament loading, the purge menu has a STOP button that doesn't work and no way to change the duration of purge.  Consider a menu item for nozzle purge that allows user to manually start/stop a purge for any amount of time between prints or during filament loading instead of a fixed duration of filament purge.

5) If you send a file to the printer using Wifi/PrusaLink while another file is printing, the currently printing file name in PrusaLink changes to the name of the sent file

thanks!  Just some ideas for consideration.

sander1095, 2023-02-05:
Hi! I am in no way affiliated with this project, but I would recommend creating seperate tickets for each of your bullet points.

This way they may be discussed individually, or closed if it is a duplicate of an existing item!

Some of these are clearly bugs and seperating them makes it easier for the devs to find and prioritize them.

I definitely agree with point 1 and 3 though! I hope that can be implemented!

Prusa-Support, 2023-03-08:
Please post every bug report or feature request as a separate issue, but first please check among the already existing issues.
If an issue already exists, please don't create a new similar one but comment in the existing thread.

Allow me to briefly comment on your points and close this issue as invalid.

1. Already existing issue.
2. I personally find it intuitive as it is now, this is a matter of personal preference though. Feel free to open an issue, but also mind this is simply a conventional choice and many customers may not appreciate a sudden change of trend
3. Already existing issue.
4. Already existing issue.
5. Already existing issue.

Michele Moramarco
Prusa Research

closed 2023-03-08 by Prusa-Support