prusa3d/Original-Prusa-MINI #76: missing filament sensor ipt files

sl1pkn07, 2024-06-16:
Hi

missing all parts of filament sensor ipt files in printed parts sources

greetings

sl1pkn07, 2025-02-26:
for the mini, they ship stp, stl and inventor(ipt) files, that files is missing

D2R-Gimli, 2025-02-26:
The ipt files are missing so native edits are not possible, yes.
You can still use the stp files as neutral cad files and edit or print them as replacement.
Years ago i've made also the full assembly based on the original parts.

[image]

sl1pkn07, 2025-02-26:
yea, but that is not the point. the point is that ipt files, like the all pieces in this project, are missing. no more, not less

the format you can use instead is irrelevant.